petergu684/HoloLens2-ResearchMode-Unity #1: Do you have any example?

mingkyme, 2020-09-09:
I tried this code.
```cs
using System;
using UnityEngine;
using UnityEngine.UI;

#if UNITY_WSA && !UNITY_EDITOR
using static HL2UnityPlugin.HL2ResearchMode;
#endif

public class SensorTest : MonoBehaviour
{
    public Text text;
#if UNITY_WSA && !UNITY_EDITOR
    HL2UnityPlugin.HL2ResearchMode researchMode;
#endif
    void Start()
    {
#if UNITY_WSA && !UNITY_EDITOR
        researchMode = new HL2UnityPlugin.HL2ResearchMode(); // System.Exception: Exception of type 'System.Exception' was thrown.
        researchMode.InitializeDepthSensor();
        researchMode.StartDepthSensorOnce();
#endif
    }
    void Update()
    {
#if UNITY_WSA && !UNITY_EDITOR
        text.text = researchMode.GetCenterDepth().ToString();
#endif
    }
}
```

Can you give me some examples?

petergu684, 2020-09-09:
Try something like this:
```(C#)
using UnityEngine;
using System;
using System.Runtime.InteropServices;
using UnityEngine.UI;

#if ENABLE_WINMD_SUPPORT
using HL2UnityPlugin;
#endif

public class SensorTest : MonoBehaviour
{
#if ENABLE_WINMD_SUPPORT
    HL2ResearchMode researchMode;
#endif

    [SerializeField]
    GameObject previewPlane = null;
    [SerializeField]
    Text text;
    private Material mediaMaterial = null;
    private Texture2D mediaTexture = null;
    private byte[] frameData = null;

    void Start()
    {
#if ENABLE_WINMD_SUPPORT
        researchMode = new HL2ResearchMode();
        researchMode.InitializeDepthSensor();
        researchMode.SetReferenceCoordinateSystem(unityWorldOrigin);
#endif
        mediaMaterial = previewPlane.GetComponent<MeshRenderer>().material;
    }

    #region Button Events
    public void PrintDepthEvent()
    {
#if ENABLE_WINMD_SUPPORT
        text.text = researchMode.GetCenterDepth().ToString();
#endif
    }

    public void PrintDepthExtrinsicsEvent()
    {
#if ENABLE_WINMD_SUPPORT
        text.text = researchMode.PrintDepthExtrinsics();
#endif
    }

    public void StartDepthSensingLoopEvent()
    {
#if ENABLE_WINMD_SUPPORT
        researchMode.StartDepthSensorLoop();
#endif
    }

    public void StopSensorLoopEvent()
    {
#if ENABLE_WINMD_SUPPORT
        researchMode.StopAllSensorDevice();
#endif
    }

    bool startRealtimePreview = false;
    public void StartPreviewEvent()
    {
        startRealtimePreview = !startRealtimePreview;
    }
    #endregion

    private void LateUpdate()
    {
#if ENABLE_WINMD_SUPPORT
        // update depth map texture
        if (startRealtimePreview && researchMode.DepthMapTextureUpdated())
        {
            if (!mediaTexture)
            {
                mediaTexture = new Texture2D(512, 512, TextureFormat.Alpha8, false);
                mediaMaterial.mainTexture = mediaTexture;
                textureInitialized = true;
            }

            byte[] frameTexture = researchMode.GetDepthMapTextureBuffer();
            if (frameTexture.Length > 0)
            {
                if (frameData == null)
                {
                    frameData = frameTexture;
                }
                else
                {
                    System.Buffer.BlockCopy(frameTexture, 0, frameData, 0, frameData.Length);
                }

                if (frameData != null)
                {
                    mediaTexture.LoadRawTextureData(frameData);
                    mediaTexture.Apply();
                }
            }
        }
#endif
    }
}
```
It is very preliminary code and may contain some bugs but I think it should be able to show the idea. Those event functions need to be assigned to buttons. You need StartDepthSensingLoopEvent before you can see the view preview.
I referred a lot to the samples in [HoloLensARToolKit](https://github.com/qian256/HoloLensARToolKit) for video preview, specifically the [ARUWPVideo](https://github.com/qian256/HoloLensARToolKit/blob/master/HoloLensARToolKit/Assets/ARToolKitUWP/Scripts/ARUWPVideo.cs). A preview plane for visualizing the depth map need to be created with the [grayscale shader](https://github.com/qian256/HoloLensARToolKit/blob/a9b8acdef53e9b5d45e70af3ea8d0c96c6e6226a/HoloLensARToolKit/Assets/Sample/Grayscale.shader). 

Could you provide more debug output if you still keep receiving the exception prompt? By the way, have you managed to run the official samples on your device?

ecde444999, 2020-09-14:
What is unityWorldOrigin?

petergu684, 2020-09-16:
Ah, it is not necessary here if you just want to show the video preview. You can simply remove that line.

petergu684, 2020-10-20:
You need to apply grayscale shader to the preview plane. [Here](https://github.com/qian256/HoloLensARToolKit) is an example showing how. You can refer to this project on how to set the preview plane. Basically, you need to create a new material in Unity and apply the [grayscale shader](https://github.com/qian256/HoloLensARToolKit/blob/master/HoloLensARToolKit/Assets/Sample/Grayscale.shader) to the material, and then apply this material to your preview plane (for example a quad).

Bornblack, 2020-11-11:
> Ah, it is not necessary here if you just want to show the video preview. You can simply remove that line.

Thank you for your great work! I finally figure out how to send point cloud data to my pc, as the point cloud in the HoloLens device Spatial Coordinate System. Now I need to align the Unity world coordinate system with the device Spatial Coordinate System. I notice the **SetReferenceCoordinateSystem(....)** method, but how to set the unityWorldOrigin?
I use this code to get unityWorldOrigin, but it seems wrong.

```C#
var WorldOrigin =(SpatialCoordinateSystem)System.Runtime.InteropServices.Marshal.GetObjectForIUnknown(
                                                          WorldManager.GetNativeISpatialCoordinateSystemPtr());

```

petergu684, 2020-11-11:
Try the following:
```cs
IntPtr WorldOriginPtr = UnityEngine.XR.WSA.WorldManager.GetNativeISpatialCoordinateSystemPtr();
var unityWorldOrigin = System.Runtime.InteropServices.Marshal.GetObjectForIUnknown(WorldOriginPtr) as Windows.Perception.Spatial.SpatialCoordinateSystem;
researchMode.SetReferenceCoordinateSystem(unityWorldOrigin);
```

agroenenberg, 2020-11-12:
@Bornblack What method did you use to get a point cloud on your pc? I added the functions, headers and soms cpp files from the StreamRecorder() sample to the winrt component but it won't fully work yet. I can get the parameters from the sensor, it makes the folder to save a file in, it just doesn't save the file. Could you give me some advice on how to get it to work?

Bornblack, 2020-11-12:
> @Bornblack What method did you use to get a point cloud on your pc? I added the functions, headers and soms cpp files from the StreamRecorder() sample to the winrt component but it won't fully work [...]

I use Peter's code to get the point cloud data and send them to PC server through TCP in **real time**. I  send the point cloud bytes to server directly without saving the point cloud data as file.
TCP Reference:
https://medium.com/datadriveninvestor/connecting-the-microsoft-hololens-and-raspberry-pi3b-58665032964c
https://foxypanda.me/tcp-client-in-a-uwp-unity-app-on-hololens/
Hope it helps!

A00107408, 2020-11-26:
How do I enable Assembly-CSharp.Player. All of the code in the #if ENABLE_WINMD_SUPPORT is greyed out.

Bornblack, 2020-11-27:
> How do I enable Assembly-CSharp.Player. All of the code in the #if ENABLE_WINMD_SUPPORT is greyed out.

Open your script in Unity, and make sure you have set Visual studio as the external editor under Preference setting. In the Visual Studio window ,there should be the **Assembly-CSharp.Playe** option under **Assembly-Csharp**. But I found the **Assembly-Csharp.Player** option is gone after my upgrading the Visual Studio. :sweat_smile:
[image]

A00107408, 2020-11-27:
I selected the box Edit -> Preferences -> External Tools -> Player Projects, then I clicked Regerenerate project files, then I got the Assembly-CSharp.Player option and not I can see all of the code in colour.

Thank you.

hein30, 2020-12-01:
@A00107408 @Bornblack What version of VS are you using? I used to have that intellisense and highlighting previously. But I upgraded my VS19 last week and now it's gone. My VS19 Community version is 16.8.2. I've regenerated project files, and even tried using Rider IDE. Am I missing something else? I still don't see Assembly-Csharp.Player in the solution explorer if that's what I need :( 

[image]

A00107408, 2020-12-01:
I'm using VS2019 but I'm dealing with some issues with this still. The settings you need are in Unity, not in VS. See the picture above my previous post.

Bornblack, 2020-12-01:
> @A00107408 @Bornblack What version of VS are you using? I used to have that intellisense and highlighting previously. But I upgraded my VS19 last week and now it's gone. My VS19 Community version is [...]

Can you find the **Assembly-CSharp.Player.csproj** file in your project root folder? 
Have you checked the **Player projects** before you Regenerate Project files? 
Maybe you should check whether the **Visual Studio Tools for unity** is installed ?

hein30, 2020-12-01:
@A00107408 @Bornblack Thank you both for your help. That was me being stupid. Didn't check the Player Projects before regenerating project files. Spent a day tweaking settings and you both saved my days. Thanks.

vbandi, 2020-12-04:
I must have spent days on this already, but for the life of me I can't get it to work. The app compiles, runs without error, but the preview is pitch black, and the depth center data remains 0. I can get the depth intrinsics though. 

Running on HoloLens 2, 10.0.19041.1128, research mode is on.

Can somebody upload a complete Unity demo project for me to try out? I must be missing a small setting somewhere, but I can't figure out where. Thank you in advance!

A00107408, 2020-12-04:
Hi vbnadi, I am not working from this repo so I have no example but I am experiencing a very similar kind of issue to what you describe. I am accessing the video camera feed and what I find is that by simply moving the line of code to display the preview it either works or doesn't work. I think this is due to blocking by non multithreaded asynchronous code in some methods. If I call my code before certain methods I have a preview, if I call it after, it seems to freeze up, in a non intuitive way. This sounds silly but try calling your preview from different places and see what happens?

vbandi, 2020-12-04:
After days of struggling, I managed to get it to work. I was using Unity 2019.4.2f1. Upgrading to 2019.4.16f1 fixed it. 

Super frustrating, but now it works.

JohannE99, 2020-12-06:
@vbandi 
Hello vbandi,
I am also trying to implement the preview currently and I am running into errors. 
Your suggestion didnt fix it for me, so it must be something else I am missing.
Could you maybe provide your working Unity Demo?
That would be super nice and helpful.
Thanks!

ThomasMountainborn, 2020-12-07:
I finally got it to work by setting the .dll version of the plugin to not be included for the WSAPlayer. I also rebuilt the .appx afterwards, though I'm not sure that that's what fixed it in the end.

JohannE99, 2020-12-17:
I also got it running now. I let the StreamRecorder with DEPTH_AHAT run beforehand and afterwards it worked (with Unity 2019.2.19f).

Bornblack, 2020-12-17:
> I also got it running now. I let the StreamRecorder with DEPTH_AHAT run beforehand and afterwards it worked (with Unity 2019.2.19f).

Try deploying the application twice! As my situation, the application with **DEPTH_LT** can work with just one deployment,  but **DEPTH_AHAT** application will work after a second deployment.

CurzeReno, 2020-12-22:
@petergu684Thanks for your excellent work！ I tried your sample code to get depth image in unity and visualize the framedata on my device. The script operates normally and sensorloop works well, but my device can't display any image. I tried to update texture using a RGB image instead, it worked, a static image can be displayed. What should I do to show the video preview, maybe framedata requires additional processing?

Bornblack, 2020-12-22:
@CurzeReno There could various reasons.
1. Make sure the **DepthMapTextureUpdated()** and **GetDepthMapTextureBuffer**() do return some values.  If it returns nothing, you need to check the inspector of .dll and .winmd files in the unity Plugins/WSA/ARM64.  Below are my setting.
[image]
[image]

 Also make sure the webcam capability is checked and you have modified the **App/[Project name]/Package.appxmanifest** correctly .
**Package.appxmanifest** :[Package.zip](https://github.com/petergu684/HoloLens2-ResearchMode-Unity/files/5728540/Package.zip)

2. If the functions mentioned above return some stuff. It could be something wrong with the shader or texture. Do you use the sample code from PeterGu? The example **Grayscale** shader takes the returned depth values as the alpha channel of the texture, and the texture background is set to black. I have done some modification to make the shader work in **Single Pass Rendering** Mode. 
Grayscale shader:[Grayscale.zip](https://github.com/petergu684/HoloLens2-ResearchMode-Unity/files/5728675/Grayscale.zip)

vbandi, 2021-01-12:
I've published a repo with a full example:
https://github.com/vbandi/HL2-ResearchMode

petergu684, 2021-03-23:
@vbandi Thanks for making this example!

closed 2021-03-23 by petergu684