petergu684/HoloLens2-ResearchMode-Unity #2: Project released

RenSiyao, 2020-10-19:
Hello, I wonder if you are still studying. I released Hololens2 according to the code in your example, but it can't work normally. Could you help me?
I have provided the background log of Hololens2, I hope you can help me look at it.
[UnityPlayer (11).log](https://github.com/petergu684/HoloLens2-ResearchMode-Unity/files/5401081/UnityPlayer.11.log)

petergu684, 2020-10-19:
Have you added the two lines of restricted capability to the package manifest (same as what you will do to enable research mode on HoloLens 1)?
For more details: http://akihiro-document.azurewebsites.net/post/hololens_researchmode2/. You just need to read the "Project settings" section to see how to modify the package manifest file (location: App/[Project name]/Package.appxmanifest).
I know it is in Japanese but you can always use right click and Translate the page if using Chrome.

RenSiyao, 2020-10-20:
> 您是否在软件包清单中添加了两行受限功能（与在HoloLens 1上启用研究模式所要做的相同）？
> 有关更多详细信息：[http](http://akihiro-document.azurewebsites.net/post/hololens_researchmode2/) : [//akihiro-document.azurewebsites.net/post/hololens_researchmode2/](http://akihiro-document.azurewebsites.net/post/hololens_researchmode2/)。您只需要阅读“项目设置”部分，即可查看如何修改软件包清单文件（位置：App / [项目名称] /Package.appxmanifest）。
> 我知道它是日语的，但是如果使用Chrome，您可以随时单击右键并翻译页面。

Thank you for your reply. I modified the project according to the link you provided, but the result after publication is still invalid.
[UnityPlayer (9).log](https://github.com/petergu684/HoloLens2-ResearchMode-Unity/files/5406023/UnityPlayer.9.log)

petergu684, 2020-10-20:
It seems that an exception is thrown when you try to initialize the instance in the Start function. The exception could be caused by many reasons. Most likely, the device is not able to obtain the Research mode functionalities or could not link to the DLL file. With so little information provided, I could hardly help you figure out the problem. Here is something you may want to check:
- Make sure you can successfully build and deploy official samples from HoloLens2ForCV repository to your device. This is to help you check whether all necessary settings are enabled.
- Build this project with ARM64, Release. Put the generated dll and winmd files into Assets/Plugins/WSA/ARM64 folder of your Unity project. When building the Unity project, please also select the build architecture to be ARM64. 
If you still have issues, please provide minimum code, versions, steps you have gone through and what exactly does it mean by "can't work normally" (e.g. Can other content of your app displayed normally or everything get stuck upon starting the app? Is there prompt asking for permission when app is started? What permissions is it asking?).

RenSiyao, 2020-10-20:
> 当您尝试在Start函数中初始化实例时，似乎引发了异常。该异常可能是由多种原因引起的。设备很可能无法获得研究模式功能或无法链接到DLL文件。由于提供的信息很少，我几乎无法帮助您解决问题。您可能需要检查以下内容：
> 
> * 确保您可以成功地从HoloLens2ForCV信息库构建正式样本并将其部署到您的设备。这是为了帮助您检查是否已启用所有必需的设置。
> * 使用ARM64，发行版构建该项目。将生成的dll和winmd文件放入Unity项目的Assets / Plugins / WSA / ARM64文件夹中。在构建Unity项目时，还请选择构建架构为ARM64。
> 
> 如果仍然有问题，请提供最低代码，版本，经过的步骤以及“不能正常工作”的确切含义（例如，应用程序的其他内容可以正常显示吗，或者启动应用程序时一切卡住了） ？启动应用程序时是否提示要求许可？它要求什么许可？）。

Thank you for your reply. I set the Unity release to ARM64 and put the DLL library in WSA/ARM64. There is no error in the project log.However, I have commented out two lines of code so far for normal operation of the project, and the depth screen in the project does not exist.
What to do to display the screen.Thank you again for your help. Thank you.
[image]

petergu684, 2020-10-20:
You need to apply grayscale shader to the preview plane. [Here](https://github.com/qian256/HoloLensARToolKit) is an example showing how. You can refer to this project on how to set the preview plane. Basically, you need to create a new material in Unity and apply the [grayscale shader](https://github.com/qian256/HoloLensARToolKit/blob/master/HoloLensARToolKit/Assets/Sample/Grayscale.shader) to the material, and then apply this material to your preview plane (for example a quad).

RenSiyao, 2020-10-21:
> 您需要将灰度着色器应用于预览平面。[这](https://github.com/qian256/HoloLensARToolKit)是显示如何的示例。您可以参考该项目，了解如何设置预览平面。基本上，您需要在Unity中创建新材质并将[灰度着色器](https://github.com/qian256/HoloLensARToolKit/blob/master/HoloLensARToolKit/Assets/Sample/Grayscale.shader)应用于该材质，然后将此材质应用于预览平面（例如，四边形）。

Thank you very much. According to what you said, I can show the screen after applying Shader and material ball.
How to use the obtained deep external events, I see that the deep external events seem to never change.
What do you use the sensor data for and in combination with other hardware?Or some other software?
Thanks again!!

petergu684, 2020-10-21:
What do you mean by deep external events? If you are able to see the depth image on the preview plane updated properly for each frame, that's it. It is up to you how to use those data in your own application. There are more data you can get from research mode. You just need to modify the `HL2ResearchMode.cpp`, `HL2ResearchMode.h` and `HL2ResearchMode.idl` files accordingly and recompile the library.

RenSiyao, 2020-10-22:
> 您所说的深层外部事件是什么意思？如果您能够在预览平面上看到针对每个帧正确更新的深度图像，就这样。如何使用这些数据取决于您自己的应用程序。您可以从研究模式中获得更多数据。你只需要修改`HL2ResearchMode.cpp`，`HL2ResearchMode.h`以及`HL2ResearchMode.idl`相应的文件，并重新编译库。您打算在C＃中公开的所有函数都应该放在`idl`文件中。

Ok, thank you very much for your reply.Thank you.

closed 2021-03-23 by petergu684