petergu684/HoloLens2-ResearchMode-Unity #4: Can not get access to IMU and LONG_THROW mode data

whirwind, 2020-10-29:
Hi @petergu684 

I am trying to extend more sensors for the project. Gary scale images works perfectly. But I can not get IMU data or LONG_THROW depth data using same code structure. The application will crash to desktop in HL2 device.

This is not about you original project, your project is great and efficient. But I still want to ask you and see if you have any advice or comment about this problem, thanks.

petergu684, 2020-10-29:
IMU data might need some extra permission as is indicated on page 18 in [official documentation](https://github.com/microsoft/HoloLens2ForCV/blob/main/Docs/ResearchMode-ApiDoc.pdf) (`<DeviceCapability Name="backgroundSpatialPerception"/>`) but Long- throw mode should be very similar to the short throw one. The difference would be that you need to use the sigma buffer to invalidate depth pixels in long throw mode. I would suggest following the example on page 12-13 of the documentation.
I also had those issues of direct crashing out of the app when I were writing the code but I do not quite remember the reason now. I would move those initializations calls you might have put in the `Start` function into different buttons can press them one by one to identify which one is causing the crash (maybe you have smarter methods).

whirwind, 2020-10-29:
You remind me about permission of IMU, that is somthing I didn't be aware of. But I did add "backgroundSpatialPerception" to the my project. There might be some other permission.

As for Long-throw mode, your are right. I know LT and ATHT and too similar so that I forgot to check buffer type. I still use 'GetBuffer()' and this might be the reason.

Thank you, I will try these solution and give you feedback

Zulex, 2020-10-29:
Please keep us up to date. I got minor experience in creating dlls from c++ code, but I'm willing to create some example scenes in Unity if you manage to read raw data in unity from the IR camera's. Let me know if I can be of any help.

whirwind, 2020-10-30:
IMU data can be accessed now. The reason is that I forget to init consent for IMU access.

But I still get app crash with LONG_THROW mode. I can get bufferSize in Unity so the thread is running. App will crash once I call 'GetDepthMapTextureBuffer()' function in Unity. I basically followed your code sturcture and the official demo so I am more confused why this happened.

Here is the code:
`

        pHL2ResearchMode->m_pLTSensor->OpenStream();
        try
        {
            while (pHL2ResearchMode->m_depthSensorLoopStarted)
            {
                IResearchModeSensorFrame* pDepthSensorFrame = nullptr;
                ResearchModeSensorResolution resolution;
                pHL2ResearchMode->m_pLTSensor->GetNextBuffer(&pDepthSensorFrame);

                pDepthSensorFrame->GetResolution(&resolution);
                pHL2ResearchMode->m_depthresolution = resolution;

                IResearchModeSensorDepthFrame* pDepthFrame = nullptr;
                winrt::check_hresult(pDepthSensorFrame->QueryInterface(IID_PPV_ARGS(&pDepthFrame)));

                size_t outBufferCount = 0;
                const BYTE* pSigma = nullptr;
                pDepthFrame->GetSigmaBuffer(&pSigma, &outBufferCount);

                const UINT16* pDepth = nullptr;
                pDepthFrame->GetBuffer(&pDepth, &outBufferCount);
                pHL2ResearchMode->m_depthbufferSize = outBufferCount;

                const UINT16* pAbImage = nullptr;
                pDepthFrame->GetAbDepthBuffer(&pAbImage, &outBufferCount);
                pHL2ResearchMode->m_AbbufferSize = outBufferCount;

                auto pDepthTexture = std::make_unique<uint8_t[]>(outBufferCount);
                auto pAbTexture = std::make_unique<uint8_t[]>(outBufferCount);

                for (UINT i = 0; i < resolution.Height; i++)
                {
                    for (UINT j = 0; j < resolution.Width; j++)
                    {
                        auto idx = resolution.Width * i + j;

                        UINT16 depth = pDepth[idx];
                        if (depth == 0) { pDepthTexture.get()[idx] = 0; }
                        else { pDepthTexture.get()[idx] = (uint8_t)((float)depth / 1000 * 255); }

                        UINT16 Ab = pAbImage[idx];
                        if (Ab == 0) { pAbTexture.get()[idx] = 0; }
                        else { pAbTexture.get()[idx] = (uint8_t)((float)Ab / 1000 * 255);}
                    }
                }

                // save data
                {
                    std::lock_guard<std::mutex> l(pHL2ResearchMode->mu);

                    if (!pHL2ResearchMode->m_depthMapTexture)
                    {
                        OutputDebugString(L"Create Space for depth map texture...\n");
                        pHL2ResearchMode->m_depthMapTexture = new UINT8[outBufferCount];
                    }
                    memcpy(pHL2ResearchMode->m_depthMapTexture, pDepthTexture.get(), outBufferCount * sizeof(UINT8));

                    if (!pHL2ResearchMode->m_AbMapTexture)
                    {
                        OutputDebugString(L"Create Space for Ab map texture...\n");
                        pHL2ResearchMode->m_AbMapTexture = new UINT8[outBufferCount];
                    }
                    memcpy(pHL2ResearchMode->m_AbMapTexture, pAbTexture.get(), outBufferCount * sizeof(UINT8));
                }

                pHL2ResearchMode->m_depthMapTextureUpdated = true;
                pHL2ResearchMode->m_AbMapTextureUpdated = true;

                pDepthTexture.reset();
                pAbTexture.reset();

                if (pDepthFrame) {
                    pDepthFrame->Release();
                }
                if (pDepthSensorFrame)
                {
                    pDepthSensorFrame->Release();
                }

            }
        }
        catch (...) {}
        pHL2ResearchMode->m_pLTSensor->CloseStream();
        pHL2ResearchMode->m_pLTSensor->Release();
        pHL2ResearchMode->m_pLTSensor = nullptr;
        pHL2ResearchMode->m_pSensorDevice->Release();
        pHL2ResearchMode->m_pSensorDevice = nullptr;
        pHL2ResearchMode->m_pSensorDeviceConsent->Release();
        pHL2ResearchMode->m_pSensorDeviceConsent = nullptr;`

petergu684, 2020-10-30:
I cannot tell the actual cause here but there are a few issues I noticed:
- You did not use the sigma buffer to check validity of pixel. In documentation, they used `0x80` mask to see whether the most significant bit is 1. Not checking that might cause problems for some invalid pixels.
- The line converting depth to grayscale value `(uint8_t)((float)depth / 1000 * 255);`: 
I was using 1000 because the valid depth pixels seems to clipped at around 1000 in short throw mode (actually it is not rigorous here as well). In long throw mode, the value can be between 200 to 7500 (mm). Using 1000 here and then cast from `float` to `int8_t` might lead to overflow issue. It applies similarly to AbImage although I am not quite sure what's the value range of that. (Probably a simple `>>8` would be fine but I haven't tested it)

Again, I do not see why these issues would lead to crash but it definitely worth a try.

whirwind, 2020-10-30:
No, I didn't use sigma buffer. Because for now the basic goal is to make it work. So I don't care about if I get correct valve yet. Theoretically I will at least get some value without any crash problem as long as the value format is right. And there do exist `(uint8_t)` to constraint it.

I tried `ConvertDepthPixel()` function from official sample to preprocess the data but it doesn't solve crash problem.
`BYTE inputPixel = ConvertDepthPixel(
                        pAbImage[resolution.Width * i + j],
                        pSigma ? pSigma[resolution.Width * i + j] : 0,
                        mask,
                        maxshort,
                        0,
                        maxClampDepth);
`

But yes, it is possible that this cause problems for some invalid pixels. I did more than one modification to check the issue, so it is hard to tell which one is the reason of the problem. I will try to preprocess these data again to see what will happen.

vbandi, 2020-12-08:
Friends, any progress on the long throw depth cam data?

ozgunkaratas, 2021-01-25:
hello 
@whirwind can you repeat how you initialized consent for the IMU? i was able to add the capablity like the following:
    <DeviceCapability Name="backgroundSpatialPerception"/>
but i was not able to initialize consent.

thank you in advance

ozgunkaratas, 2021-01-26:
hello!
were you able to obtain information from the IMU and save it to some folder?

JaVo93, 2021-04-14:
> Friends, any progress on the long throw depth cam data?

I am also interested are any new versions with long throw depth cam data?

made3, 2021-04-15:
@whirwind
Do you mind sharing your working IMU code?

closed 2021-11-15 by petergu684