petergu684/HoloLens2-ResearchMode-Unity #5: Cannot get GetCenterPoint and GetCenterDepth to work...

vbandi, 2020-11-27:
Hi,

First of all, thank you very much for this. 

I'm trying to take small steps, and first trying to get the center point depth displayed somehow. I managed to get the little sample from #1 compiled, and tweaked it a little so that I don't even use any visual display except for the "text".

The HL2ResearchMode gets loaded, and I can print the extrensics just fine. The StartDepthSensorLoop is also called without exception. However, when I try to update the label with either 
`        var cp = researchMode.GetCenterPoint();        
        text.text = $"{cp[0]}, {cp[1]}, {cp[2]}";`

or just 
`text.text = researchMode.GetCenterDepth().ToString() `

the output remains "0,0,0" or simply "0" in every frame.

Also, PrintResolution logs "0x0x0", which feels strange :)

Any help would be greatly appreciated.

petergu684, 2020-11-27:
It sounds strange that you are able to print out the extrinsic but fail to get the other data. There is one [known issue](https://github.com/microsoft/HoloLens2ForCV/issues/20#issuecomment-693442823) about research mode that if you start the app with AHAT camera functionalities and the HoloLens somehow gets into hibernation later, then you cannot get the depth sensor data after you wake it up. You have to restart the device to let it work again. I am not sure whether this solves your problem but you definitely may want to try restart the device.

vbandi, 2020-11-28:
Thank you for the answer. I have tried restarting the device, but it doesn't help. I've also made sure that the call to StarTDepthSensorLoop happens 2 seconds after the rest of the initialization to avoid any race conditions. (I'm not even using buttons yet). 

HL OS is latest publicly available one, Unity version is 2019.4.2f1.

vbandi, 2020-12-04:
After days of struggling, I managed to get it to work. I was using Unity 2019.4.2f1. Upgrading to 2019.4.16f1 fixed it.

Super frustrating, but now it works.

closed 2020-12-04 by vbandi