petergu684/HoloLens2-ResearchMode-Unity #6: How to debug within this project

zhangyk18, 2020-11-30:
Thansk very much for your repository. I have succeed deploy the ResearchMode-Unity. But the debug process is really inconvenient. So do you have any good suggestions for this?

Any replay will be appreciated.

petergu684, 2020-12-01:
Yes, I also find it not easy to debug the C++ part of the code. What I did was using the `OutputDebugString` function to print some intermediate results. That will be printed in visual studio output window. I kept some of them in the code as well. Please let me know if you have smarter ways of debugging. XD

zhangyk18, 2020-12-01:
The way I am using is defining the debug information as the output of the functions, like `return` or `out`, that's too complicated. Maybe I can try the `OutputDebugString` you refered.
Hopefully there will be fewer bugs in the program. XD
Once again thank you.

closed 2021-03-23 by petergu684