petergu684/HoloLens2-ResearchMode-Unity #7: Flipped point cloud

hein30, 2020-12-01:
Thank you very much for your repo. It's amazingly easy to follow and to extend. 

I am using it to generate point clouds for LT sensor. I have managed to make it all work, but the final point clouds are flipped left/right as shown below. I haven't used the `SetReferenceCoordinateSystem` yet. And the z values seem about right too. 
In the image, I have my left hand raised and it's looking like a right hand. 
Thanks again for your amazing work.
[image]

petergu684, 2020-12-01:
How did you save the point cloud? What transforms did you applied? If you followed exactly the same as I did on AHAT depth data, then it is saved in Unity left-handed coordinate system (because I want to visualize it in Unity). Then it makes sense that point cloud is flipped if you visualize the point cloud on your laptop with software of right-handed system. You can remove the negative sign [here](https://github.com/petergu684/HoloLens2-ResearchMode-Unity/blob/4ff374d2a80a24ca40491cd4892c06cbb3beda31/HL2UnityPlugin/HL2ResearchMode.cpp#L181) to change it back to right-handed.

hein30, 2020-12-01:
@petergu684 Thanks a lot for your reply. It was indeed the negative sign 👍

closed 2020-12-01 by hein30

michalek-marcin, 2020-12-16:
@hein30 can you share what you've changed in HL2ResearchMode and how you use it in Unity project to get point cloud data from LT sensor?