petergu684/HoloLens2-ResearchMode-Unity #8: DepthMapTextureUpdated() always returns false

BradenLuancingC, 2020-12-10:
I'm using your implementation for testing, and without modifying anything in the C# sample script or the C++ files. My plane wasn't displaying depth information, and when I did some debugging to check, the check in SensorTest.cs in the main Update() loop:
`if (startRealtimePreview && researchMode.DepthMapTextureUpdated())`

The latter is always false. Are there any obvious reasons that may cause this, like the HL2 is configured wrong, or is the problem in the C++ likely in the code? I did test with default HL2ForCV samples to make sure my ResearchMode works correctly and is enabled.

Thanks for any help.

 Unity v.2020.1.14f1
VS2019 v16.8.0

BradenLuancingC, 2020-12-11:
An update on the issue: this only seems to happen on the first run of the program when it asks for permissions. For whatever reason, I have to restart the Hololens, then the app works fine. Is this related to the closing/opening of streams maybe?

petergu684, 2020-12-11:
I don't know why it only happens on the first run which I apparently do not have this issue. However it is indeed a known issue that if you open the sensor stream and your HoloLens later enters hibernation mode after being inactive for several minutes, you have to restart the device or otherwise the sensor data will not be available. This is the case even I close the sensor stream after each use before exiting the app. (Maybe I did not do it correctly)

wanglolly, 2021-01-11:
Hi @petergu684! Thank you for this nice work!
I also did the same implementation and find that the code `DepthMapTextureUpdated()` always returns false.
The code `StartDepthSensingLoopEvent()` and `StartPreviewEvent()` are called after 3 seconds in Unity. 
I also restart the Hololens2 after building the project, but it seems that the code always return a null pointer in the C++ code ` auto transToWorld = pHL2ResearchMode->m_locator.TryLocateAtTimestamp(ts, pHL2ResearchMode->m_refFrame);`
And therefore the code `DepthMapTextureUpdated()` returns false.
Do you have any idea or suggestion for it? 

Thank you for your help!

Unity v.2019.4.17f
VS2019 v16.8.3

wanglolly, 2021-01-14:
Just update the issue for those who might have the same problem in the future :)
This is quite a silly one. 
I solve it after I import the MRTK plugin into the Unity program.
Before importing the MRTK plugin, the unity program was a 2D app running in Hololens, and therefore the 3D transform to the world was not available.
Still, thanks for the great work!

closed 2021-03-23 by petergu684

GLARKI, 2022-11-06:
Could you please sent me a link to the correct repo? 
With me the problem is very persistent, restarts and reattempts do not work. Maybe I'm missing something obvious though, any help is appreciated:

My source:

https://github.com/microsoft/HoloLens2ForCV/tree/main/Samples/StreamRecorder