petergu684/HoloLens2-ResearchMode-Unity #10: How to build a c++dll?

JiangTao100, 2021-03-01:
@petergu684 I wanna build a c++ dll ,but I find there may be something wrong with my dll. I just built a simple function for ARM64 to deploy it to hololens2.
I just delete all your HL2RmStreamUnityPlugin .h  and .cpp  code and write a return function.
The .h file is like this:

`#pragma once

#ifdef FUNCTIONS_EXPORTS  
#define FUNCTIONS_EXPORTS_API extern "C" __declspec(dllexport)   
#else  
#define FUNCTIONS_EXPORTS_API extern "C" __declspec(dllimport)   
#endif  

namespace HL2Stream
{
FUNCTIONS_EXPORTS_API   int __stdcall DOUT();
}`
And the .cpp file is below:
`#include "pch.h"
#include "HL2RmStreamUnityPlugin.h"

#define DBG_ENABLE_VERBOSE_LOGGING 1
#define DBG_ENABLE_INFO_LOGGING 1

 int __stdcall HL2Stream::DOUT()
{
	return 123;
}`
then I deploy it into Unity/Plugin and the scritp is:
`[DllImport("HL2RmStreamUnityPlugin.dll", EntryPoint = "DOUT", CallingConvention = CallingConvention.Cdecl)]
    public static extern int DOUT();
`

But  i find there is something wrong with it. 
First of all, I guess  it is the version problem.
I find your project .vcxproj version 
`    <WindowsTargetPlatformVersion>10.0.18362.0</WindowsTargetPlatformVersion>
    <WindowsTargetPlatformMinVersion>10.0.17763.0</WindowsTargetPlatformMinVersion>`
and my unity build setting is 
[image]
Next,  maybe there are more detailed settings in building dll. And i would be appreciated
that someone can help ！

JiangTao100, 2021-03-01:
Hololens2: 10.0.19041.1136
Unity:  2019.4.12f1
VS: 2019

petergu684, 2021-03-23:
Sorry for the late reply. I am using the Windows Runtime Extension template in this repository but what you are doing seems to be related to building native plugins for Unity. There are a lot of tutorials online teaching you how to build the native plugins in Unity (for example, if you wanna use OpenCV). I suggest you create a new project following those tutorials rather than using this project. The native plugins won't have the winmd file, and you need to do DLLImport in the script.