petergu684/HoloLens2-ResearchMode-Unity #11: Get raw RGB Frame

pallavimohansaab, 2021-03-10:
Is it possible to get the raw RGB/PV camera frames through this .dll? 

In the original C++ app, it saves the RGB frames along with the AHAT results, and I'm looking to extract the raw RGB frame in Unity as well. However, I didn't find a suitable function in the built .dll. Any pointers?

Thanks in advance!

petergu684, 2021-03-23:
You don't need to enable research mode to access the PV camera. You can check [this script](https://github.com/qian256/HoloLensARToolKit/blob/master/HoloLensARToolKit/Assets/ARToolKitUWP/Scripts/ARUWPVideo.cs) how to obtain and visualize PV camera frames in Unity.

kaiwu119, 2021-04-22:
Hello, I try to obtain a pv image based on the point cloud to get a color point cloud. I use [PhotoCapture](https://forum.unity.com/threads/holographic-photo-blending-with-photocapture.416023/) directly obtains the pv image in unity/C#, there is no problem when this method is run alone. However, when I used PhotoCapture and research mode at the same time, some abnormalities occurred. The point cloud was continuously acquired, and the acquisition of the pv image was controlled by the button. When four to five pv images are obtained, the program exits directly. And when I run the program again, the research mode API seems to be unable to obtain the deep point cloud information, and I must restart hololens. Perhaps this is a conflict between the pv camera and the depth camera?
Is there any possible reason or solution?
Any help would be greatly appreciated.

petergu684, 2021-05-15:
I don't think PhotoCapture is designed for video streaming tasks as its name indicates. [This Project](https://github.com/doughtmw/HoloLensForCV-Unity) has the implementation to map RGB frame to point cloud on HoloLens 1. Probably that would be useful to you.

shoukailiang, 2022-01-07:
> I don't think PhotoCapture is designed for video streaming tasks as its name indicates. [This Project](https://github.com/doughtmw/HoloLensForCV-Unity) has the implementation to map RGB frame to point cloud on HoloLens 1. Probably that would be useful to you.

Hello,there are any Hololens2 solutions to achieve video sreaming tasks in unity ?