petergu684/HoloLens2-ResearchMode-Unity #12: How to make the code available in my project?

Scrooge96, 2021-03-22:
Hi, I tried using your project (the compiled .dll and .winmd) in my existing project. I followed the steps you described exactly but the packages couldn't be imported into my code files (using command failed). Also multiple regenerations of the code assembly didn't change this. Is there a need to manually link the dlls in the assembly project? I tried linking the .dll but got an error that it could not be added since it is neither accessible nor a valid assembly nor a COM component. Since I am not too familiar with VS and you did not mention any of this in your guide I stopped checking on this. Any hint?

petergu684, 2021-03-23:
I don't think you need to link the dlls manually in the Unity project. And it is not possible for me to find out the issue if you only tell me "using command failed". One thing to mention is that all the functions in dll should be wrapped inside `#if ENABLE_WINMD_SUPPORT` in your Unity scripts. I might be able to help look into the problem if you can provide more specific information about the errors and the related settings of your project.

Scrooge96, 2021-03-23:
Sorry about being unspecific. The problem is the IDE (VS19) claims to not find a package with this name (HL2UnityPlugin). Therefore the autocompletetion also does not work. Another problem is that with my unity project type set correctly (UWP > ARM64) `ENABLE_WINMD_SUPPORT`  is not true when editing the code and also not when compiling the project. Thus the code after `#if ENABLE_WINMD_SUPPORT` is not executed. How did you expect my project to be set up already before importing your plugin for this to work?

petergu684, 2021-03-23:
May be check the settings here?
https://docs.unity3d.com/Manual/IL2CPP-WindowsRuntimeSupport.html

Scrooge96, 2021-03-23:
Thanks, I already checked on this. I also experimented quite a bit with the plugin settings in the unity editor although I think this is not intended. No change in the described behaviour. Problem remains.

Scrooge96, 2021-03-25:
I don't think that the compiled files are the problem and assume that won't help much but I would like to try: Could you make your working compiled .dll and .winmd files available for download maybe.

petergu684, 2021-03-25:
Try [this one](https://github.com/petergu684/HoloLens2-ResearchMode-Unity/releases) maybe?

closed 2021-11-15 by petergu684