petergu684/HoloLens2-ResearchMode-Unity #13: How to modify in order to access IMU data?

made3, 2021-04-16:
Hey,

first of all: Thank you very much for this project!
I do not need the depth camera though, but the IMU values. I am not sure how to achieve this though, so I tried accessing a test function first. So I did the following steps:
1) Added a function `void TestResearchmodeFunction();` into the `HL2ResearchMode.idl`
2) Added the same function to the `HL2ResearchMode.h`
3) Added an implementation of the function in the `HL2ResearchMode.cpp` (just calculating 2+2 or whatever, no return value of course)
4) Built the solution successfully
5) Reimported the plugins (dll and winmd) in my Unity project successfully
6) Built the Unity project and the output solution successfully with all manifest settings etc correctly
7) Deployed and started it

Now a console output in the app on the HoloLens 2 shows up telling me "invalid cast exception specified cast is not valid"
Is this the correct way to add a function in your project or did I miss a step? 

Btw: Before adding my test function everything worked well and I was able to initialize the depth sensor and read its center value etc. I just have problems after adding a function to the .idl, .h and .cpp, which I thought was everything I would need to do.

Thank you already in advance!

petergu684, 2021-04-16:
Hi Martin, what you did should be correct. The cast exception seems to be related to what you actually put in your implementation function. Would you mind sharing the test function you used that introduced this exception?

made3, 2021-04-16:
> Hi Martin, what you did should be correct. The cast exception seems to be related to what you actually put in your implementation function. Would you mind sharing the test function you used that introduced this exception?

Thank you so much for your fast reply! I am not in my office until monday, so I can't look it up. But it was not much. It was seriously just:
adding `void TestResearchmodeFunction()` into the `HL2ResearchMode.idl`
adding `void TestResearchmodeFunction()` into the `HL2ResearchMode.h`
and something like `void HL2ResearchMode::TestResearchmodeFunction() { 
int i = 1;
for(int j = 0; j < 5; j++){
    i *= j;
}
return; 
}` 
into the `HL2ResearchMode.idl`

So the actual function did not really do anything.

Just to see what happens I also simply commented out the `GetCenterDepth()` function in all 3 files and it brought up the same error on the HoloLens. Obviously I did not use the function in my Unity project, otherwise it would not have built it.

It's like, if all functions are the same as in your project it works, but as soon as a function is added or removed it won't work anymore.

made3, 2021-04-19:
The issue is solved! Or well, I am fairly certain that it was the fix at least. I noticed that I have never set the import settings of the plugins, so for the HL2UnityPlugin.dll I have now set the SDK to UWP and the CPU to ARM64. For the HL2UnityPlugin.winmd I have just set the SDK to UWP.
Not sure though why that would be the solution because I was able to get the DepthCenter before as well, but I won't complain.

@petergu684 Do you have the same import settings?

petergu684, 2021-04-19:
Thanks for the update and glad you figured that out. But for me, I never manually changed those import settings in Unity so I don't know what's going on and why your app has this behavior.

Sayoj, 2021-05-05:
@made3 Hey Martin! 

I just started to implement the IMU capability to the code and saw your thread. 
Would you mind sharing your code with me?

Regards

made3, 2021-05-05:
@Sayoj Hey! 
I would love to! I always thought about sharing my code here but I forgot.
Gotta do that tomorrow though as I am not on my work PC anymore. :)

made3, 2021-05-06:
@Sayoj These are the files, I don't think anything else had to be changed. I think the accelerometer could cause some problems, at least in one Unity application it's not working, but I did not check it again since then. 
We can not receive the calibrated magnetometer values, just the raw values. So I have done a little calibration, but it is not tilt compensated yet. So if your head is straight it works alright overall (Though the values jump quite a bit because the calibration is not very good I guess). But if you tilt your head it won't work. I think a Kalman Filter is needed.

Anyway, you can get the degree starting from north clockwise in what direction you are looking (`GetMagDegrees`) (When starting you should first make a full 360 degree rotation with the headset to get the maximum and minimum values) and you could get the direction (`GetMagDirection`) ( North, East etc)

If you have any questions feel free to ask, because once I stop working on this project I will forget all of it.

[HL2ResearchModeFiles.zip](https://github.com/petergu684/HoloLens2-ResearchMode-Unity/files/6432360/HL2ResearchModeFiles.zip)

Sayoj, 2021-05-06:
@made3 Thank you very much for the files! 

I'll look into it right now :) 

Do you also have a example script for unity to use it?

made3, 2021-05-06:
@Sayoj 

I think this should work: It takes 3 TextMeshPro objects to write the values to. First you have to use the StartIMULoops() function (For example with a button event) and then it will read and show the values gathered from all 3 IMU sensors. It does not read the raw magnetometer values, but just the degrees and direction, but I think you could read the raw data as well.

```
using System.Collections;
using System.Collections.Generic;
using UnityEngine;
using TMPro;

#if ENABLE_WINMD_SUPPORT
using HL2UnityPlugin;
#endif

public class ResearchModeTest : MonoBehaviour
{
#if ENABLE_WINMD_SUPPORT
    HL2ResearchMode researchMode;
#endif

    [SerializeField]
    TextMeshPro accel;
    [SerializeField]
    TextMeshPro gyro;
    [SerializeField]
    TextMeshPro mag;

    // Start is called before the first frame update
    void Start()
    {
#if ENABLE_WINMD_SUPPORT
        researchMode = new HL2ResearchMode();
        researchMode.InitializeAllSensors();
#endif
    }

    public void StartIMULoops()
    {
#if ENABLE_WINMD_SUPPORT
        researchMode.StartAccelSensorLoop();
        researchMode.StartGyroSensorLoop();
        researchMode.StartMagSensorLoop();
#endif
    }

    // Update is called once per frame
    void LateUpdate()
    {
#if ENABLE_WINMD_SUPPORT
        string tmpString = "X: " + researchMode.GetAccelValues()[0].ToString("F1") + "\nY: " + researchMode.GetAccelValues()[1].ToString("F1") + "\nZ: " + researchMode.GetAccelValues()[2].ToString("F1");
        accel.text = tmpString;
        tmpString = "X: " + researchMode.GetGyroValues()[0].ToString("F1") + "\nY: " + researchMode.GetGyroValues()[1].ToString("F1") + "\nZ: " + researchMode.GetGyroValues()[2].ToString("F1");
        gyro.text = tmpString;
        tmpString = researchMode.GetMagDirection() + " " + researchMode.GetMagDegrees();
        mag.text = tmpString;
#endif
    }
}
```

Sayoj, 2021-05-12:
@made3 

Thank you for sharing all those lines of code! It really helped me a lot. 
Everything worked as expected.

Only thing I have to add for other potential users: 
Don't forget to add `<DeviceCapability Name="backgroundSpatialPerception"/>` to Package.appxmanifest

Regards
Sayoj

made3, 2021-05-20:
@Sayoj 

Is still everything working without any problems? And what sensors of the three do you use?

I run into app crashes from time to time caused by the magnetometer sensor loop...

Sayoj, 2021-05-21:
Hey @made3 

Yes, everything is working just fine! Thanks for the update, I will integrate it.
I just need the Accelerometer and Gyroscope data. But it is not final, so I kept the magnetometer in for possible use in the future. 

Nevertheless I've got one question: 
What is the original variable of the Values? As for now, I would like to get them in double. 
I can convert the string into double, but why the extra step, if I know the original variable. 

Thanks and best regards!

made3, 2021-05-25:
Hey @Sayoj 

What do you mean by that? In Unity you receive the Accelerometer and Gyroscope data as an array of floats, not as a string. And the original value you get from the sensor is also a float.

You could try to change all `com_array<float>` to a `com_array<double>` in all `GetXValues()` functions. No idea if it works that easily.

axteroide, 2021-06-08:
Hi @made3 

I've been trying to make you code work unsuccessfully. I tried the base project and worked fine.
Then I changed the DLL for the DLL I generated with the files in the zip you uploaded "HL2ResearchModeFiles.zip".
Then I removed the ResearchModeVideoStream.cs and added your ResearchModeTest.cs
Compiled Unity proyect and added the backgroundSpatialPerception device capability to the manifest, then compiled to the Holo 2.
I only get X: 0 Y: 0 Z: 0
and "dir not found 0" for the mag sensor.

Any ideas of what did I do wrong?

Thank you!

made3, 2021-06-09:
Hey @axteroide 

I can't really tell. I think I never used the base project, so I am not sure what you replaced with my test-script. But I can give a few hints that might be the problem?

For the manifest, I had to change two spots. (Not sure anymore why I had to change them though)
This is my second line of the manifest:
```
<Package xmlns="http://schemas.microsoft.com/appx/manifest/foundation/windows10" xmlns:mp="http://schemas.microsoft.com/appx/2014/phone/manifest" xmlns:uap="http://schemas.microsoft.com/appx/manifest/uap/windows10" xmlns:uap2="http://schemas.microsoft.com/appx/manifest/uap/windows10/2" xmlns:rescap="http://schemas.microsoft.com/appx/manifest/foundation/windows10/restrictedcapabilities" xmlns:wincap="http://schemas.microsoft.com/appx/manifest/foundation/windows10/windowscapabilities" IgnorableNamespaces="uap uap2 mp rescap">
```
And these are the capabilities:
```
  <Capabilities>
    <uap:Capability Name="enterpriseAuthentication" />
    <Capability Name="privateNetworkClientServer" />
    <Capability Name="internetClient" />
    <Capability Name="internetClientServer" />
    <uap:Capability Name="documentsLibrary" />
    <uap2:Capability Name="spatialPerception" />
    <rescap:Capability Name="perceptionSensorsExperimental" />
    <DeviceCapability Name="microphone" />
    <DeviceCapability Name="gazeinput" />
    <DeviceCapability Name="webcam" />
    <DeviceCapability Name="wifiControl" />
    <DeviceCapability Name="backgroundSpatialPerception" />
  </Capabilities>
```

You probably won't need all of them, but you could try changing those two manifest spots and see if you get results then. If yes, you can strip them down to the ones that caused the problem.

In my application sometimes a developer console popped up while using the HoloLens 2. To this day I have no idea why that was the case because I never really activated any developer console, but it showed me error messages. Maybe look for such a console on the bottom left of your HoloLens screen? 

Last thing I would suggest is to use `OutputDebugString` in your `HL2ResearchMode.cpp`. To be honest, I don't remember if it worked, but I think that when you printed something with the `OutputDebugString` it also printed it in Visual Studio if you started the app with the VS debugger. That way you could check if you get correct values from the HoloLens, which means that the problem is not a permission problem but something within the Unity Project.

Try to understand how the code works and ask if you have any other questions!

Sayoj, 2021-06-11:
@made3 
Hey sorry for the late reply! 

Yes, I wanted to know if it's a float or not ^^

axteroide, 2021-06-17:
Thank you for your answer @made3 

Just doing some tests I just realized on the Unity script sample you posted here you included el method StartIMULoops()
But it was never called!! After adding a call to the method all worked smoothly, so luckily in the end it was just a silly mistake I did not realizing that call was missing...

Thanks again!

axteroide, 2021-07-05:
Hey @made3 sorry for bothering you again.

Now that I got access to the magnetometer I've been struggling to understand the data but I get some strange things. I don't really have much experience with IMU data so maybe you can help me with this.
My intentions are to position things in the real world with GPS coordinats so for that I need to get at least the angle of the hololens for Y coordinate in Unity, kind of what I get with GetMagDegrees() but I don't think this data is reliable enough It's more or less well if you are looking forward but as soon as you look up and down I get a 10º-20º error.
If I read the raw data with GetMagValues() I get kind of strange values like X: 100 Y: 420 z: -90 and rotating around the axis doesn't seem to get the full 360º, like, if I turn 360º around my head Y axis, maybe I get a varianz fo -40 to +40 of some axis and I quit don't get how the values are changing.
The same way you get magValueY and magValueZ I tried to get magValueX and display it but again, in this case, I don't see values from -PI to PI or from 0 to 2PI.

I'm also thinking about implementing a Kalman Filter to see if things improve, I'll let you know the results I get.

Regards!

made3, 2021-07-05:
Hey @axteroide 

Yes, so the values for Accelerometer and Gyroscope are calibrated. The values that the Research Mode gives for the Magnetometer are not.
The raw data you get from it are probably Tesla values, but no one actually knows. And for me they were like x:200 y: 400 z:-200 or so (I am not sure anymore, but they were all high values!). So I did a hard iron and soft iron filtering to bring them down to be around 0.

> If I read the raw data with GetMagValues() I get kind of strange values like X: 100 Y: 420 z: -90 and rotating around the axis doesn't seem to get the full 360º, like, if I turn 360º around my head [...]

These values are Teslas or something like that. And the -40 and +40 is kind of the up and down range it has. So basically if your max X is 140 and your min X is 60 you take the mean value of these two which is (obviously) 100. Now you have your offset. So to get the values without hard iron distortion you calculate X - offsetX. That would give you an X between -40 and 40. Now you can normalize that value so that it is between 0 and 1 or whatever value you like.

Kalman Filter will probably work, but it was too much for me to implement. I would like to hear your approach though.

And for the GetMagDegrees(), to get more reliable values you have to "calibrate" the HoloLens by just randomly rotating it in all directions. That way it will get the highest maximum and lowest minimum value for each axis. They are sadly still not perfect. And it honestly bothers me that Microsoft only gives us the raw Tesla values and a shitty/non-existing documentation instead of calibrated data and a description of what we are actually looking at.

linngn, 2021-09-09:
Hi @made3,
I used your .zip file HL2ResearchModeFiles.zip to build .dll file but idk why I got this error:
_1>C:\Program Files (x86)\Windows Kits\10\bin\10.0.18362.0\XamlCompiler\Microsoft.Windows.UI.Xaml.Common.targets(482,5): error MSB4181: The "CompileXaml" task returned false but did not log an error.
1>Done building project "HL2UnityPlugin.vcxproj" -- FAILED._
How can I solve this?
thankyouu

made3, 2021-09-09:
> Hi @made3,
> I used your .zip file HL2ResearchModeFiles.zip to build .dll file but idk why I got this error:
> _1>C:\Program Files (x86)\Windows [...]

Hey,
I gotta be honest, it was way too long ago since I worked with it. So I have no idea. Did you have the same problems when building the base files? The one from this github repository? (https://github.com/petergu684/HoloLens2-ResearchMode-Unity)

linngn, 2021-09-11:
@made3 I deployed successfully from your mentioned repo, but when i tried to modify his code by your .zip file then that error came out. I think i missed some steps to compile or sth T.T
Btw, i used this repo and this work:  https://github.com/petergu684/HoloLens2-ResearchMode-Unity/pull/17

made3, 2021-09-11:
> @made3 I deployed successfully from your mentioned repo, but when i tried to modify his code by your .zip file then that error came out. I think i missed some steps to compile or sth T.T
> Btw, i used this repo and this work: #17

So the problem is solved? Isn't the work you mentioned kind of doing the same as I did?

linngn, 2021-09-13:
> So the problem is solved? Isn't the work you mentioned kind of doing the same as I did?

yep, but I stuck in deploying .dll in VS2019 and got that error.

linngn, 2021-09-23:
Sorry for bothering you again because of my stupid question because I'm new to programming T.T
I added a void test function simply like this:
`void testingSomething(float a)
    {
        a = a + 1;
        return;
    }`
So I added in .cpp, .h, and .idl files, but VS2019 announced an error and cannot build the solution.
Do I need to do some steps after added the function in those files?
Thanks in advance.

petergu684, 2021-11-15:
Feature merged (#17 )

closed 2021-11-15 by petergu684