petergu684/HoloLens2-ResearchMode-Unity #14: Error while getting point cloud

kaiwu119, 2021-04-18:
Hello, thank you for sharing this project. I currently have a problem. After I compiled the library, I tried to get the point cloud data in Unity. At first, I tried to output the depth center, it always returned 0, but no error occurred. Then I use the same method to output the number of point clouds. At this time, when running in hololens2, it will prompt: NullReferenceException: Object reference not set to an instance of an object.
Excuse me, is there any way to solve my problem.
My code as follow:
`using System.Collections;
using System.Collections.Generic;
using UnityEngine;
using System;
using System.Runtime.InteropServices;
using TMPro;
using UnityEngine.UI;

#if ENABLE_WINMD_SUPPORT
using HL2UnityPlugin;
#endif
public class SensorTest : MonoBehaviour
{
#if ENABLE_WINMD_SUPPORT
    HL2ResearchMode researchMode = null;
#endif

   // [SerializeField]
    //GameObject previewPlane = null;
    //[SerializeField]
    public TextMeshPro text;
    private Material mediaMaterial = null;
    private Texture2D mediaTexture = null;
    private byte[] frameData = null;

    private bool isDepthLoop = false;
    // Start is called before the first frame update
    void Start()
    {

#if UNITY_WSA && !UNITY_EDITOR
        researchMode = new HL2UnityPlugin.HL2ResearchMode(); // System.Exception: Exception of type 'System.Exception' was thrown.
        researchMode.InitializeDepthSensor();
        text.text = "init DepthSensor ok ";
#endif
    }

    // Update is called once per frame
    void Update()
    {

        if (!isDepthLoop) StartDepthSensingLoopEvent();
#if UNITY_WSA && !UNITY_EDITOR
        var pointCloud = researchMode.GetPointCloudBuffer();
        text.text = " points count "+ (pointCloud.Length / 3).ToString();
        text.text += "myTest " + researchMode.GetCenterDepth().ToString();
#endif
    }
    public void StartDepthSensingLoopEvent()
    {
#if ENABLE_WINMD_SUPPORT
        IntPtr WorldOriginPtr = UnityEngine.XR.WSA.WorldManager.GetNativeISpatialCoordinateSystemPtr();
        var unityWorldOrigin = System.Runtime.InteropServices.Marshal.GetObjectForIUnknown(WorldOriginPtr) as Windows.Perception.Spatial.SpatialCoordinateSystem;
        researchMode.SetReferenceCoordinateSystem(unityWorldOrigin);
        researchMode.StartDepthSensorLoop();
        text.text = "start Loop ";
#endif
        isDepthLoop = true;
    }

}
`
Any help would be greatly appreciated.

kaiwu119, 2021-04-19:
I found that I can get the point cloud data after restarting Hololens, but during the acquisition process, Hololens console still shows NullReferenceException: Object reference not set to an instance of an object. But the program runs fine. What is the problem?

petergu684, 2021-04-19:
Yes, it is an issue that you need to restart HoloLens if the HoloLens enter sleeping mode before you stop the sensor stream properly and I still haven't look into how to solve it elegantly. 
As for the NullReferenceException, can you locate the line that triggers this exception? Is it in update function that keeps showing up or does it just happens at the very first one or few iterations?

kaiwu119, 2021-04-20:
Sorry for the late reply. I found that this kind of exception is occasionally reported when the point cloud data stream is continuously obtained. This kind of anomaly can only be seen in hololens and is displayed in red. But this kind of anomaly does not affect the point cloud acquisition process, but it is a little uncomfortable.I have not been able to pinpoint exactly where the error occurred,  I will continue to debug to find the place where the exception occurred.

petergu684, 2021-05-15:
I uploaded a Unity sample that includes visualizing point cloud. Could you try whether that sample has similar behavior on your device?

EugeneVasiliev, 2021-06-22:
The error when accessing the depth sensor buffer after waking up from sleep is a [known system problem](https://github.com/microsoft/HoloLens2ForCV/issues/10) and is fixed when the OS is updated to **20346.1403**. arm64fre.fe_release_svc_sydney.210518-1040 (or later).

At the moment, the specified OS build is in the Beta channel ([Insider preview](https://docs.microsoft.com/en-us/hololens/hololens-insider))

After the update, your plugin works like a charm, thanks for the work!

closed 2021-11-15 by petergu684