petergu684/HoloLens2-ResearchMode-Unity #16: Can not get the depth image through TCP Client/Server

suzi-soohyun, 2021-07-19:
Hello everybody!

I tested the TCP client/server codes which are recently you added.
But the problem is, I can not see any depth image in pc through TCP server.
It's just a 'data' folder and no files in there when I running the Hololens app. (Sensor streaming is alright.)

Did I miss any steps? Or should I edit some parts of the scripts?
If anybody knows the solution, please let me know. Thank you!

petergu684, 2021-07-20:
You need to change the ip address in the TCPClient script in Unity to your laptop's IP address and make sure your laptop and HoloLens are in the same network. There should be some message printed on the python side once it receives images from HoloLens.

suzi-soohyun, 2021-07-20:
@petergu684 
[image]

I can see the connected message but can not save the images in the "data" folder.
Is this a network problem also?

petergu684, 2021-07-20:
Then the network should be fine. On the HoloLens side, you need to click Space button to send images to laptop (see KeyboardCommand.cs). You can do this through Device Portal, or connect a wireless keyboard to HoloLens, or add a button and set the onClick event to the specific function you want to call.

suzi-soohyun, 2021-07-21:
[image]

Virtual keyboard in the Device portal was not working so I made a button for the function call to save the depth images.
And now it worked! Thank you so much :)

closed 2021-07-21 by suzi-soohyun

AnandMulay, 2022-12-22:
@suzi-soohyun did you changed anything in the Python file, I just added the localhost IP in python and added my laptop IP in unity with the same port 9090 but the connection is not happening