petergu684/HoloLens2-ResearchMode-Unity #18: Long throw sensor Point Cloud?

vbandi, 2021-08-03:
Hi,

I'd like to request a way to get the point cloud for the long throw sensor. I had a look at HL2ResearchMode::DepthSensorLoop, and I think I found where to do it, but I'm not a C++ dev and can barely understand what's going on. The code seems to be written around the short throw sensor, and thus modifying it to include the long throw sensor for point cloud is beyond what I'm able to do. 

I understand that the key is to use MapImagePointToCameraUnityPlane, which also exists for the long depth sensor.

Zulex, 2021-09-25:
@vbandi 
Did you manage to do this? I have the exact same issue.

vbandi, 2021-09-25:
@Zulex nope.

petergu684, 2021-11-15:
I may try to do it later but I suppose it should be pretty similar to the short throw one?

petergu684, 2021-11-17:
Added :)

closed 2021-11-17 by petergu684

vbandi, 2021-11-17:
@petergu684 yohooo! I'll check it soon.

vbandi, 2021-12-10:
Thanks! Works well, submitted a PR with a small fix. I see some crashes in my app when turning off the depth sensor altogether, but couldn't find the cause yet.

Sindhu2k3, 2024-10-17:
> Hi,
>
> I'd like to request a way to get the point cloud for the long throw sensor. I had a look at HL2ResearchMode::DepthSensorLoop, and I think I found where to do it, but I'm not a C++ dev and can [...]

I am facing the same issue. if the problem is solved could you please guide me through it