petergu684/HoloLens2-ResearchMode-Unity #19: pointcloud transformed with a constant positional and rotational offset

marcfischer123467, 2021-08-16:
Thank you for this amazing project! 
I have a quick question: When I use the unity sample, the point cloud of objects within 1m distance is often transformed with a constant positional and rotational offset that sometimes changes after stopping the stream using the provided UI and closing/restarting the app. After several iterations of stopping the stream and closing/restarting the app, the problem goes away but seems to come back randomly. Otherwise, it works perfectly fine.. ?
Could someone advise where I'm going wrong? The transform from the depth sensor to the rig node in HL2ResearchMode.cpp makes sense I think and no other transforms are introduced in ResearchModeVideoStream.cs or PointCloudRenderer.cs. I feel like I'm missing something obvious? I'll keep debugging and make a more formal report but I just wanted to reach out beforehand to see if someone has had the same issue.

petergu684, 2021-11-15:
How large the offset could be? I know there is several centimeter offset in the depth value for those point cloud but in plane position (x-y) looks fine on my device. 
Maybe try the following to obtain the unity origin?
```cs
IntPtr WorldOriginPtr = UnityEngine.XR.WindowsMR.WindowsMREnvironment.OriginSpatialCoordinateSystem;
var unityWorldOrigin = Marshal.GetObjectForIUnknown(WorldOriginPtr) as Windows.Perception.Spatial.SpatialCoordinateSystem;
```

luohwu, 2022-08-16:
Hi mrcfschr, @mrcfschr 

I hope that you are doing well. Did you solve this problem? I encounter the same problem. Would you mind sharing your solution? Thanks in advance!

mrcfschr, 2022-08-16:
we did solve it but I'm unfortunately not allowed to share our code since it's for a company :/ It's definitely very possible

luohwu, 2022-08-16:
Hi mrcfschr, 
Thanks for your reply. I just solved it. Turned out that my collegue modified `unityWorldOrigin` and `researchMode.SetReferenceCoordinateSystem(unityWorldOrigin);` into a wrong value.

For people who might encounter the same problem in the future:
make sure you configure them correctly accoridng to your unity version.