petergu684/HoloLens2-ResearchMode-Unity #21: Can I get RGB-D data with this repo?

euijaebilly, 2021-09-23:
Thank you very much for your hard work.

I want to get RGB-D data using HoloLens 2.
In the official repo, separately stored data is processed using Python and then data is created.
But I want to get RGB-D data from the device and store it frame by frame.

The most similar thing I found is
https://github.com/cyberj0g/HoloLensForCV
https://github.com/doughtmw/HoloLensForCV-Unity

euijaebilly, 2021-09-28:
What I need is to make the depth data displayed as a curved surface into a flat image, and find a region within it that matches the color image.

petergu684, 2021-11-15:
Mapping between RGB image and depth image is not implemented in this repo. The second repo you mentioned (https://github.com/doughtmw/HoloLensForCV-Unity) have the implementation you need on HoloLens 1. Probably you could refer to that and try to do that yourself :)

closed 2021-11-15 by petergu684