petergu684/HoloLens2-ResearchMode-Unity #25: IMU Sensor Data Loop Errors Out

Deltajom, 2021-11-16:
Hello, I am currently working on a class project involving the Hololens 2 @ University and have run into an issue. I have tried to use your dll and WINMD files with some success, but unfortunately, I am unable to get any data from the sensors I specify. I use your unity example code as the base, and have the following settings for my project...

Unity Version - 2019.4.22f1
MRTK Version - 2.7.2
Minimum Platform Version - 10.0.18362.0
Editor - VS 2019 (Latest installed in Build Settings)
Api Compatibility Level - .Net Standard 2.0
Allow 'unsafe' code - True
Plugin Providers - Windows Mixed Reality checked
Depth Buffer - 16 Bit

I know I have research mode working because I am able to get the HoloLensCV2 sample projects working, and I've followed the instructions specified you have provided so I am unsure of what would be next. I am able to build and load the app, however, and the other functionality for my app seems to be fine. I was finally able to attach a debugger and it pointed me to the accelerometer, gyro, and magnetometer sensor loops, I have included some images below.

ERROR GIVEN
[image]

Line of code that threw above error
[image]

I will be working on this for the next ~month or so, but at this point, I will also try to modify the HoloLens2CV sample code for my needs as I also need to start the paper.

Any help or suggestions would be appreciated, I can give any extra output if needed. If you would be willing to post what exact settings you are able to run the IMU sensor sample app with I would be greatful. :smile:

petergu684, 2021-11-16:
Could you post your `Package.appxmanifest` file?

Deltajom, 2021-11-16:
Sure here you go

<?xml version="1.0" encoding="utf-8"?>
<Package xmlns:mp="http://schemas.microsoft.com/appx/2014/phone/manifest" xmlns:uap="http://schemas.microsoft.com/appx/manifest/uap/windows10" xmlns:uap2="http://schemas.microsoft.com/appx/manifest/uap/windows10/2" xmlns:uap3="http://schemas.microsoft.com/appx/manifest/uap/windows10/3" xmlns:uap4="http://schemas.microsoft.com/appx/manifest/uap/windows10/4" xmlns:iot="http://schemas.microsoft.com/appx/manifest/iot/windows10" xmlns:mobile="http://schemas.microsoft.com/appx/manifest/mobile/windows10" xmlns:rescap="http://schemas.microsoft.com/appx/manifest/foundation/windows10/restrictedcapabilities" IgnorableNamespaces="uap uap2 uap3 uap4 mp mobile iot rescap" xmlns="http://schemas.microsoft.com/appx/manifest/foundation/windows10">
  <Identity Name="Template3D" Publisher="CN=DefaultCompany" Version="1.0.0.0" />
  <mp:PhoneIdentity PhoneProductId="xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx" PhonePublisherId="00000000-0000-0000-0000-000000000000" />
  <Properties>
    <DisplayName>697FinalTryUnity</DisplayName>
    <PublisherDisplayName>DefaultCompany</PublisherDisplayName>
    <Logo>Assets\StoreLogo.png</Logo>
  </Properties>
  <Dependencies>
    <TargetDeviceFamily Name="Windows.Universal" MinVersion="10.0.18362.0" MaxVersionTested="10.0.22000.0" />
  </Dependencies>
  <Resources>
    <Resource Language="x-generate" />
  </Resources>
  <Applications>
    <Application Id="App" Executable="$targetnametoken$.exe" EntryPoint="Template3D.App">
      <uap:VisualElements DisplayName="697FinalTryUnity" Square150x150Logo="Assets\Square150x150Logo.png" Square44x44Logo="Assets\Square44x44Logo.png" Description="Template_3D" BackgroundColor="transparent">
        <uap:DefaultTile ShortName="697FinalTryUnity" Wide310x150Logo="Assets\Wide310x150Logo.png" />
        <uap:SplashScreen Image="Assets\SplashScreen.png" BackgroundColor="#FFFFFF" />
        <uap:InitialRotationPreference>
          <uap:Rotation Preference="landscape" />
          <uap:Rotation Preference="landscapeFlipped" />
          <uap:Rotation Preference="portrait" />
          <uap:Rotation Preference="portraitFlipped" />
        </uap:InitialRotationPreference>
      </uap:VisualElements>
    </Application>
  </Applications>
  <Capabilities>
    <rescap:Capability Name="perceptionSensorsExperimental" />
    <Capability Name="internetClient" />
    <uap2:Capability Name="spatialPerception" />
    <uap3:Capability Name="backgroundMediaPlayback" />
    <DeviceCapability Name="webcam" />
    <DeviceCapability Name="backgroundSpatialPerception"/>
    <DeviceCapability Name="microphone" />
    <DeviceCapability Name="gazeinput" />
  </Capabilities>
</Package>

Deltajom, 2021-11-16:
``` xml
<?xml version="1.0" encoding="utf-8"?>
<Package xmlns:mp="http://schemas.microsoft.com/appx/2014/phone/manifest" xmlns:uap="http://schemas.microsoft.com/appx/manifest/uap/windows10" xmlns:uap2="http://schemas.microsoft.com/appx/manifest/uap/windows10/2" xmlns:uap3="http://schemas.microsoft.com/appx/manifest/uap/windows10/3" xmlns:uap4="http://schemas.microsoft.com/appx/manifest/uap/windows10/4" xmlns:iot="http://schemas.microsoft.com/appx/manifest/iot/windows10" xmlns:mobile="http://schemas.microsoft.com/appx/manifest/mobile/windows10" xmlns:rescap="http://schemas.microsoft.com/appx/manifest/foundation/windows10/restrictedcapabilities" IgnorableNamespaces="uap uap2 uap3 uap4 mp mobile iot rescap" xmlns="http://schemas.microsoft.com/appx/manifest/foundation/windows10">
  <Identity Name="Template3D" Publisher="CN=DefaultCompany" Version="1.0.0.0" />
  <mp:PhoneIdentity PhoneProductId="xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx" PhonePublisherId="00000000-0000-0000-0000-000000000000" />
  <Properties>
    <DisplayName>697FinalTryUnity</DisplayName>
    <PublisherDisplayName>DefaultCompany</PublisherDisplayName>
    <Logo>Assets\StoreLogo.png</Logo>
  </Properties>
  <Dependencies>
    <TargetDeviceFamily Name="Windows.Universal" MinVersion="10.0.18362.0" MaxVersionTested="10.0.22000.0" />
  </Dependencies>
  <Resources>
    <Resource Language="x-generate" />
  </Resources>
  <Applications>
    <Application Id="App" Executable="$targetnametoken$.exe" EntryPoint="Template3D.App">
      <uap:VisualElements DisplayName="697FinalTryUnity" Square150x150Logo="Assets\Square150x150Logo.png" Square44x44Logo="Assets\Square44x44Logo.png" Description="Template_3D" BackgroundColor="transparent">
        <uap:DefaultTile ShortName="697FinalTryUnity" Wide310x150Logo="Assets\Wide310x150Logo.png" />
        <uap:SplashScreen Image="Assets\SplashScreen.png" BackgroundColor="#FFFFFF" />
        <uap:InitialRotationPreference>
          <uap:Rotation Preference="landscape" />
          <uap:Rotation Preference="landscapeFlipped" />
          <uap:Rotation Preference="portrait" />
          <uap:Rotation Preference="portraitFlipped" />
        </uap:InitialRotationPreference>
      </uap:VisualElements>
    </Application>
  </Applications>
  <Capabilities>
    <rescap:Capability Name="perceptionSensorsExperimental" />
    <Capability Name="internetClient" />
    <uap2:Capability Name="spatialPerception" />
    <uap3:Capability Name="backgroundMediaPlayback" />
    <DeviceCapability Name="webcam" />
    <DeviceCapability Name="backgroundSpatialPerception"/>
    <DeviceCapability Name="microphone" />
    <DeviceCapability Name="gazeinput" />
  </Capabilities>
</Package>
```

petergu684, 2021-11-16:
Sorry I am not able to provide extra insights. Does the imu sample scene work for you? The error seems to be complaining not able to get imu data and usually it is something related to privilege settings but it doesn't seems to be your case.

Deltajom, 2021-11-16:
Yeah I gave the sample scene a try but wasn't getting any output from that either, would you mind describing your exact unity setup? (Build settings, unity version, MRTK version, Api Compatibility Level, and Plugin Provider used (OpenXR vs UWP)?)

petergu684, 2021-11-16:
My settings are similar to yours. But the MRTK I was using is 2.6.1. All my settings should be saved in the ProjectSettings folder and MRTK version can be found in Packages folder. 
What you really need to do is to "Add" project in Unity Hub and select the UnitySample folder as the project folder. Then change the platform to UWP and open the project. Packages will be auto imported and setting will be applied according to the files in ProjectSettings folder.
After all packages are imported, open one of the two scenes and you are good to go.

Deltajom, 2021-11-16:
Some success! I was able to get the point cloud visualizer to load and run, but the IMU sample scene still gave me no output in the text field. (The app runs and I can see the white box for accel, gyro, and mag, but it is otherwise blank and never updates.) This is also after one restart to accept all the permissions of course. I'll keep trying while I work to see but I'm curious about the sensor loop, could they have changed something in either the platform version that is breaking it?

Deltajom, 2021-11-17:
HUGE UPDATE - I got IMU to work. You need to set minimum platform version to 10.0.15063.0 or it won't give any data. Thanks for creating this, and for responding @petergu684 😄 😄 😄