petergu684/HoloLens2-ResearchMode-Unity #31: PointCloud offset when used with FrozenWorld 1.5.7 (Unity 2019.4.22f1) World Anchor WSA

chezulkhairi, 2022-01-31:
PointCloud "Origin offset" when used with FrozenWorld 1.5.7 (Unity 2019.4.22f1) World Anchor WSA.

I have disabled this line #researchMode.SetReferenceCoordinateSystem(unityWorldOrigin); (ResearchModeVideoStream.cs) - hoping that the PointCloud PCL rendering will not be based on a different (origin 0,0,0) position+orientation. 
However, with FrozenWorld 1.5.7 - the Spongy (origin 0,0,0) is following where you restart HL2 app, while the F1(is fixed to where you first launch the Unity App). So, most of the time Spongy (origin 0,0,0) and F1(origin 0,0,0) position and orientation is not the same. 
Following this, the PCL will be rendered with similar offset values.
If Spongy (origin 0,0,0) and F1(origin 0,0,0) is offset by x,y,z and rx,ry,rz rotation, then the PCL is rendered in that similar position and orientation.
Is there to make sure that the PCL rendering switch-off its own (origin) and use FrozenWorld origin, instead?

Thanks, Zul

chezulkhairi, 2022-02-01:
Hi Peter, 
I can parent all the pointcloud and later on move the parent to the Frozen/world spongy origin, and rotate the axis as well. And, check if there is no differences in the both origin(PCL's parent and spongy- stop this activity)
However, I am hoping for a different way of achieving this, so that I can use FrozenWorld WSA anchor with the PCL rendering at the right spot. and angle.

camnewnham, 2022-05-20:
I encountered this as well. In our case we resolved it by transforming each point in the cloud with:
```C#
    // PointCloudRenderer
    public void Render(Vector3[] arrVertices)
    {
        Pose spatialTransform = WorldLockingManager.GetInstance().FrozenFromSpongy;
        for (int i = 0; i < arrVertices.Length; i++)
        {
            arrVertices[i] = spatialTransform.Multiply(arrVertices[i]);
        }
    }
```
Note that in this scenario there are no changes to precompiled dll, and the point cloud renderer is _not_ parented by the `MixedRealityPlayspace` or `AdjustmentNode`