petergu684/HoloLens2-ResearchMode-Unity #33: IMU Sensor Frequency of acquisition

ZAO29, 2022-02-03:
Hello,
Just a simple question, I tried the Unity Scene "ImuViewSample". I would like to acquire the IMU sensor to a certain frequency (for example  60 Hz). It seems that the sensor values are updated  less quickly than this. Is it  a limitation of the sensor speed or a limitation of the C++ function HL2ResearchMode::AccelSensorLoop in  HL2UnityPlugin sln ?

ZAO29, 2022-02-03:
Another question, how can we know the units of measurement used by the output of the IMU sensors (Magneto, Accelerometer,  Gyroscope) ? I didn't find any documentation ...

FlorianDirnberger, 2023-07-04:
> Another question, how can we know the units of measurement used by the output of the IMU sensors (Magneto, Accelerometer, Gyroscope) ? I didn't find any documentation ...

As far as i found out it is m/s rad/s and microtesla (not 100% sure about the last one)