petergu684/HoloLens2-ResearchMode-Unity #35: Things that don't work in this project.

arsi0001, 2022-02-10:
First big thanks to the developer for the project! I noticed that a few things do not work and I hope to get help here.

So I tried this project today and right now its a bit frustrating.
I got it running on the HoloLens 2 and the python server is also running. 

So what is my problem?

1. After the application started on the HoloLens and the connection to the python server is established, I can click "save spatial image" and it works! If I do so, the server recognize the command under ` if header == 'f': # save spatial camera images` and saves it on under `/data`
2. If I click then on "save ahat" and then again on "save spatial image" the server get the command to trigger  `if header == 's': # save depth sensor images`
3. Regardless of the order in which something is clicked, the AHAT command never works! Nothing happens on the python server side.
4. As already someone mentioned on the issue page, the stop button causes the application to crash.

arsi0001, 2022-02-11:
added more information in my first post

arsi0001, 2022-02-11:
does someone know why GetDepthMapBuffer returns null?

arsi0001, 2022-02-15:
I found the solution.
Short: The wrong depth sensors where initialized and the c++-code had no error handling for that.
Long: The problem was that `DepthSensorMode `was serialized and visible in the unity editor. So depthSensorMode was in the editor view set to `LongThrow `and the whole code was expecting `ShortThrow ` what you set at the beginning of the class ResearchModeVideoStream with `[SerializeField] DepthSensorMode depthSensorMode = DepthSensorMode.ShortThrow;`. So the depth sensor for long was initialized and `SaveAHATSensorDataEvent()` got for `depthMap` and `AbImage = null`.

arsi0001, 2022-02-16:
@petergu684 @HoloAdventure
So there is another problem with reshaping the point cloud data. I added the information to the first post!

arsi0001, 2022-02-17:
I fixed my reshape problem by realizing that `var depthMap = researchMode.GetDepthMapBuffer();` is not the method for that. And the method `SendUINT16Async()` with one parameter is also not the right method for that, so I coded another one.

```
    public void SavePointCloud()
    {
#if ENABLE_WINMD_SUPPORT
        var pointCloudBuffer = researchMode.GetPointCloudBuffer();
#if WINDOWS_UWP
        if (tcpClient != null)
        {
            tcpClient.SendSingleAsync(pointCloudBuffer );
        }
#endif
#endif
    }
```

```
    bool lastMessageSent = true;
    public async void SendSingleAsync(Single[] data)
    {
        if (!lastMessageSent) return;
        lastMessageSent = false;
        try
        {
            // Write header
            dw.WriteString("p"); // header "p" 

            // Write point cloud
            dw.WriteInt32(data.Length);
            dw.WriteBytes(SingleToBytes(data));

            // Send out
            await dw.StoreAsync();
            await dw.FlushAsync();
        }
        catch (Exception ex)
        {
            SocketErrorStatus webErrorStatus = SocketError.GetStatus(ex.GetBaseException().HResult);
            Debug.Log(webErrorStatus.ToString() != "Unknown" ? webErrorStatus.ToString() : ex.Message);
        }
        lastMessageSent = true;
    }
```

agroenenberg, 2022-07-22:
@arsi0001 , thanks so much for your contribution! I have one more question, since I still got the problem that the point cloud is not received properly. I think I wrote the SingleToBytes(); function correctly, but I can't think of another place where the bug could be. How did you solve the problem? Here's my code:

    byte[] SingleToBytes(float[] data)
    {
        byte[] floatInBytes = new byte[data.Length * sizeof(float)];
        System.Buffer.BlockCopy(data, 0, floatInBytes, 0, floatInBytes.Length);
        return floatInBytes;
    }

arsi0001, 2022-07-25:
@agroenenberg 
Hey, your method should be fine, mine looked the same:
`    byte[] SingleToBytes(Single[] data)
    {
        byte[] singleInBytes = new byte[data.Length * 4];
        System.Buffer.BlockCopy(data, 0, singleInBytes, 0, singleInBytes.Length);

        return singleInBytes;
    }`
    
    What is your error message? Or what's wrong with the point cloud you received?

agroenenberg, 2022-08-03:
Hey @arsi0001, I've been going through it and it seems that no matter what I do, the maximum size of `data` is never more than 12288, while the length of `singleInBytes` is definitely about +- 3 times more than that and also prints the correct length just before sending in Unity... I just can't seem to figure out why the received `data` is so different, while `data.Length`(C#) or `N_pointcloud` does print the right length. Maybe something is wrong on the python side?

dhb123, 2022-09-02:
> I found the solution.
> 
> **Short**: The wrong depth sensors where initialized and the c++-code had no error handling for that.
> 
> **Long**: The problem was that `DepthSensorMode` was serialized and visible in the unity editor. So depthSensorMode was in the editor view set to `LongThrow ` (by default) and the whole code was expecting `ShortThrow` what you set at the beginning of the class ResearchModeVideoStream with `[SerializeField] DepthSensorMode depthSensorMode = DepthSensorMode.ShortThrow;`. So the depth sensor for long was initialized and `SaveAHATSensorDataEvent()` got for `depthMap` and `AbImage = null`.

Hi, thanks so much for your contribution! @arsi0001 .However, I am a newcomer in hololens and I do not quite understand your solution about the AHAT. Can you elaborate on `Short: The wrong depth sensors where initialized and the c++-code had no error handling for that.`?Thank you!

Yuhan-Shen, 2022-09-12:
> I found the solution.
>
> **Short**: The wrong depth sensors where initialized and the c++-code had no error handling for that.
>
> **Long**: The problem was that `DepthSensorMode` was serialized and [...]

Hi, I can use `researchMode.InitializeLongDepthSensor(); researchMode.StartLongDepthSensorLoop(enablePointCloud); researchMode.GetLongDepthMapBuffer()` to get the long_throw depth images. But after I set `[SerializeField] DepthSensorMode depthSensorMode = DepthSensorMode.ShortThrow;` I am unable to access the short_throw depth images by `researchMode.InitializeDepthSensor(); researchMode.StartDepthSensorLoop(enablePointCloud); researchMode.GetDepthMapBuffer()`. The error is the same as yours (`var depthMap = researchMode.GetDepthMapBuffer()`;  returns null). Do you have any idea why this happens and how I can get the short_throw depth images? Thanks a lot!

arsi0001, 2022-09-15:
@Yuhan-Shen
If all lines in the code are set to shortThrow, it should work. Like I already said, I got that error only because there was a hardcoded line for a different depthSensoreMode.

Do you rely on the shortThrow? Because the shortThrow has a bug anyway, that is caused by the original researchmode project.

Yuhan-Shen, 2022-09-15:
> @Yuhan-Shen If all lines in the code are set to shortThrow, it should work. Like I already said, I got that error only because there was a hardcoded line for a different depthSensoreMode.
>
> Do you [...]

Thank you! I really appreciate your reply.

To clarify, is the depthSensorMode set only in Unity/C#. Do I need to check the C++ plugins?

Also, can you please let me know what is the bug for shortThrow? Thanks!

arsi0001, 2022-09-19:
> To clarify, is the depthSensorMode set only in Unity/C#? Do I need to check the C++ plugins?
No, no need to check there. The only thing missing in c++ is the handling for the case of wrong depth mode. So if you initialize for short but then ask for the long depth data.

[Here](https://github.com/microsoft/HoloLens2ForCV/issues/12 )is the bug.