petergu684/HoloLens2-ResearchMode-Unity #40: namespace can not be found in unity script

mahdidBagheri, 2022-04-02:
[image]

Hi. Thank you for good repo.
I want to use this plugin in my own project, but script in unity can not find plugin namespace even though I followed the instruction (builed in Release/ARM64 copied .dll & .winmd to Assets/Plugins/WSA/ARM64). did I missed sth or it needs further configuration?

closed 2022-05-23 by mahdidBagheri

anjelomerte, 2024-02-16:
Hey, curious how you got it working. Currently trying to figure that out as well. Any help much appreciated:)

mahdidBagheri, 2024-02-16:
Actually I really can't remember. I think I didn't. Just cloned the repo and trimmed unwanted parts and developed based on it. :)