petergu684/HoloLens2-ResearchMode-Unity #44: researchMode object seems to be null

yogesh-sriraman, 2022-08-22:
Hi!
I am trying to build point cloud sample in Unity. My Editor version is 2020.3.34f1.
I followed the README file in the github page to deploy the build in Hololens 2.
I have also enabled research mode on my hololens.
However, when I run the code, I was getting a Null reference exception.
Few debugs later, I found out that `researchMode` seems to be null in `LateUpdate()` of `ResearchModeVideoStream`.

I also checked the permissions: Camera, Microphone and Eye tracker as given. I reopened the app in hololens.
I also tried restarting hololens. I still get the NullReferenceException issue when I call `researchMode.LongDepthMapTextureUpdated()`.

Is there any step that I missed? I can also provide with more information if needed.

Thank you!

arsi0001, 2022-08-22:
I guess that happens because the depth variant (maybe short depth) is hardcoded in the class, but you try to use long depth. Thats something that I experienced [here](https://github.com/petergu684/HoloLens2-ResearchMode-Unity/issues/35#issuecomment-1040335196).

yogesh-sriraman, 2022-08-22:
I made the code changes as well.
Went to apps and provided permissions to Camera, Microphone, Eye gaze.
Restarted the hololens again.
The problem still seems to exist. I am getting the same NullReference error

yogesh-sriraman, 2022-08-23:
I tried to open the project in 2019.4.28f1.
Everything else is the same.
It is working now.
Not sure why it didn't work in 2020.3.
Anyway, a good start to learn research mode.
I can work on top of it now.

Thank you!

closed 2022-08-23 by yogesh-sriraman