petergu684/HoloLens2-ResearchMode-Unity #46: How to make it works in unity and show in Hololens 2?

Hans86, 2022-09-08:
Thanks for author's project.  When i try to follow the steps to put the .dll and .winmd into my unity project, it didn't show the depth camera view in hololens.My hololens just show the view that i built before, but there no any error msg, what's wrong with this situation, do i need to do anything to activate the .dll?

arsi0001, 2022-09-12:
what is your dll-path in your project?
do you see the dll in the builded project in the solution explrorer of visual studio?
did you enable developer mode and research mode on you hololens?
did you set the capabilities the right way?
starting the application the first time, do you get the typical notifications/questions about capabilities?

You are providing no information...

Hans86, 2022-09-12:
I built author's .sln to get the .dll and .winmd, then I put them into my unity project, and the path is `Assets/Plugins/WSA/ARM64.`
[image]
I'm sure that I turn on the mode because I have run the Microsoft sample project and it can be run.
After I built my project, it looks success and there's no any error notification.
[image]
[image]
Should I change any setting about these two file?Thanks for your help.

arsi0001, 2022-09-12:
Those settings are ok.
What version of unity are you using?
What XR Plugin-in Managment are you using? (Windows Mixed Reality or OpenXR)

Hans86, 2022-09-13:
The version I use is 2020.3.14f, I try to use the version that author provide has these problem.
[image]

My XR Plugin-in Management  is OpenXR and feature group is set to Microsoft Hololens like this.
[image]

arsi0001, 2022-09-13:
You have to work with Windows Mixed Reality. Remove OpenXR.

Hans86, 2022-09-13:
I tried to use Windows Mixed Reality,but it still not work.
[image]

After put two data in Plugins, should I do anything else?

Appreciate to your reply, I'll try to find wether the path is too long.

arsi0001, 2022-09-14:
if you followed the instruction for this plugin, there nothing else you have to do. It should work.
Did you setup the manifest file the right way?

Hans86, 2022-09-15:
I think I followed the instruction, is it possible to have your successful project, I don't know what happen to my project.